zephyrproject-rtos/west #724: error: https://github.com/NordicDeveloperAcademy/ncs-inter did not send all necessary objects

olalonde, 2024-08-11:
I keep getting this error. Any idea what's wrong? Tried deleting `/opt/nordic/ncs/v2.7.0/nrf/samples/devacademy/ncs-inter/.git/` and trying again but keeps failing.

```sh
$ west update -x devacademy-ncsinter                                          v2.7.0 ✗ 
WARNING: ignoring --exclude-west
=== updating devacademy-ncsinter (nrf/samples/devacademy/ncs-inter):
--- devacademy-ncsinter: initializing
Initialized empty Git repository in /opt/nordic/ncs/v2.7.0/nrf/samples/devacademy/ncs-inter/.git/
--- devacademy-ncsinter: fetching, need revision main
remote: Enumerating objects: 509, done.
remote: Counting objects: 100% (104/104), done.
remote: Compressing objects: 100% (49/49), done.
remote: Total 509 (delta 63), reused 55 (delta 55), pack-reused 405
Receiving objects: 100% (509/509), 106.40 KiB | 1.40 MiB/s, done.
Resolving deltas: 100% (240/240), done.
usage: git rev-list [<options>] <commit-id>... [-- <path>...]
  limiting output:
    --max-count=<n>
    --max-age=<epoch>
    --min-age=<epoch>
    --sparse
    --no-merges
    --min-parents=<n>
    --no-min-parents
    --max-parents=<n>
    --no-max-parents
    --remove-empty
    --all
    --branches
    --tags
    --remotes
    --stdin
    --quiet
  ordering output:
    --topo-order
    --date-order
    --reverse
  formatting output:
    --parents
    --children
    --objects | --objects-edge
    --unpacked
    --header | --pretty
    --[no-]object-names
    --abbrev=<n> | --no-abbrev
    --abbrev-commit
    --left-right
    --count
  special purpose:
    --bisect
    --bisect-vars
    --bisect-all
error: https://github.com/NordicDeveloperAcademy/ncs-inter did not send all necessary objects

ERROR: update failed for project devacademy-ncsinter
/opt/nordic/ncs/v2.7.0/nrf/samples/devacademy/ncs-inter0 rm -rf .git                                                         init_placeholder 
/opt/nordic/ncs/v2.7.0/nrf/samples/devacademy/ncs-inter$ west update -x devacademy-ncsinter                                          v2.7.0 ✗ 
WARNING: ignoring --exclude-west
=== updating devacademy-ncsinter (nrf/samples/devacademy/ncs-inter):
--- devacademy-ncsinter: initializing
Initialized empty Git repository in /opt/nordic/ncs/v2.7.0/nrf/samples/devacademy/ncs-inter/.git/
--- devacademy-ncsinter: fetching, need revision main
fatal: unable to access 'https://github.com/NordicDeveloperAcademy/ncs-inter/': Could not resolve host: github.com
ERROR: update failed for project devacademy-ncsinter
/opt/nordic/ncs/v2.7.0/nrf/samples/devacademy/ncs-inter0 rm -rf .git; west update -x devacademy-ncsinter                     init_placeholder 
WARNING: ignoring --exclude-west
=== updating devacademy-ncsinter (nrf/samples/devacademy/ncs-inter):
--- devacademy-ncsinter: initializing
Initialized empty Git repository in /opt/nordic/ncs/v2.7.0/nrf/samples/devacademy/ncs-inter/.git/
--- devacademy-ncsinter: fetching, need revision main
remote: Enumerating objects: 509, done.
remote: Counting objects: 100% (104/104), done.
remote: Compressing objects: 100% (49/49), done.
remote: Total 509 (delta 63), reused 55 (delta 55), pack-reused 405
Receiving objects: 100% (509/509), 106.40 KiB | 131.00 KiB/s, done.
Resolving deltas: 100% (240/240), done.
usage: git rev-list [<options>] <commit-id>... [-- <path>...]
  limiting output:
    --max-count=<n>
    --max-age=<epoch>
    --min-age=<epoch>
    --sparse
    --no-merges
    --min-parents=<n>
    --no-min-parents
    --max-parents=<n>
    --no-max-parents
    --remove-empty
    --all
    --branches
    --tags
    --remotes
    --stdin
    --quiet
  ordering output:
    --topo-order
    --date-order
    --reverse
  formatting output:
    --parents
    --children
    --objects | --objects-edge
    --unpacked
    --header | --pretty
    --[no-]object-names
    --abbrev=<n> | --no-abbrev
    --abbrev-commit
    --left-right
    --count
  special purpose:
    --bisect
    --bisect-vars
    --bisect-all
error: https://github.com/NordicDeveloperAcademy/ncs-inter did not send all necessary objects

ERROR: update failed for project devacademy-ncsinter
```

marc-hb, 2024-08-12:
As shown in the logs, the `-x` is ignored now. Please stop using `-x`, it's confusing.

Please share complete reproduction steps so anyone can reproduce from scratch.

In the initial description there was `Could not resolve host: github.com`. Do you have unreliable Internet?

The `rev-list` error looks like something bad in the `west.yml`, please check it.

`west update -v ...` is also useful.

marc-hb, 2024-08-15:
Let's re-open once you have more data.

closed 2024-08-15 by marc-hb