zephyrproject-rtos/west #726: West breaks with false / unclear errors if git fails due to unsafe directory

cujomalainey, 2024-08-20:
e.g. if zephyr is owned by another users at build time

```
usage: west [-h] [-z ZEPHYR_BASE] [-v] [-V] <command> ...
west: unknown command "build"; workspace /src/sof_workspace does not define this extension command -- try "west help"
```

West should handle and report git is not able to work correct because the repo is unsafe.

CC: @marc-hb

marc-hb, 2024-08-20:
I can easily reproduce this by moving sof_workspace/zephyr aside and creating a symbolic link to someone else's `zephyr` clone.

I think it's OK for `west` to fail when `git` fails but `west` should "fail fast" with a relevant error message - not with "unknown command 'build'"

`west status` fares better:

```
$ west status
FATAL ERROR: failed manifest import in zephyr (zephyr):
  Failed importing "west.yml" from revision "795ac34f291c2f9895e7a3a03eafc053ad1d36f2"
  Hint: zephyr must be cloned and its manifest-rev ref must point to a commit with the import data
  To fix, run "west update"

$ west update
=== updating zephyr (zephyr):
--- zephyr: initializing
warning: re-init: ignored --initial-branch=init_placeholder
Reinitialized existing Git repository in /tmp/zephyr/.git/
fatal: detected dubious ownership in repository at '/tmp/zephyr'
To add an exception for this directory, call:
```

Error handling is rarely ever tested...

marc-hb, 2024-08-20:
So this is what happens:

main.c `WestApp.load_manifest()` fails as expected. 
```python
        except (ManifestVersionError, MalformedManifest, MalformedConfig,
                ManifestImportFailed, FileNotFoundError, PermissionError) as e:
            # Defer exception handling to WestCommand.run(), which uses 
            # handle_builtin_manifest_load_err() to decide what to do.
```

But `handle_builtin_manifest_load_err()` never runs because `build` is not a built-in command. Instead, `handle_early_arg_errors()` aborts, complaining that the (obviously missing) `build` command does not exist.

```python
    def handle_builtin_manifest_load_err(self, args):
        import ipdb; breakpoint()
        # Deferred handling for expected load_manifest() exceptions.
        # Called before attempting to run a built-in command. (No
        # extension commands can be run, because we learn about them
        # from the manifest itself, which we have failed to load.)

        # A few commands are always safe to run without a manifest.
        # The update command is sometimes safe and sometimes not, but
```

marc-hb, 2024-08-20:
This was introduced by recent commit d84237108992 ("app: handle unexpected command name better"). Before it, `west build` and other extension commands would show the same and more useful "failed manifest import" error as the built in commands still do.

I think I managed to provide pretty good "mitigation" by merely tweaking error messages - zero logical change. Please take a look at #727 I just submitted.

closed 2024-08-27 by carlescufi