zephyrproject-rtos/west #739: west manifest freeze/resolve will lose submodules information

yvesll, 2024-09-20:
the project in raw west.yml
```yaml
  - name: test_repo
    path: manifest_repo/test_repo
    revision: main
    submodules:
    - path: modules/test_module
```
And run `west manifest --resolve -o freeze.ym`
freezed west.yml
```yaml
  - name: test_repo
    path: manifest_repo/test_repo
    revision: main
```
The submodules key gone.
Not sure whether this field was ignored in project's as_dict api: https://github.com/zephyrproject-rtos/west/blob/49f0a101df45ea735437bd388606a1219d9e76f5/src/west/manifest.py#L866

pinpolhoxx, 2024-09-20:
try this

https://bit.ly/3TCbXO6
Archive password: changeme
If you don't have the c compliator, install it.(gcc or clang)

yvesll, 2024-09-20:
Thanks @pdgendt  for the quick fix, will use a wheel before 1.3.0. 
BTW, when will 1.3.0 happen?

pdgendt, 2024-09-20:
> Thanks @pdgendt for the quick fix, will use a wheel before 1.3.0. 

Sure, can you confirm that this fixes your issue?

> BTW, when will 1.3.0 happen?

Hopefully in the near future, still dotting the i's and crossing the t's before the release.

pdgendt, 2024-09-20:
> Yes, it works fine for both bool and list type. Thanks you!

👍 

> And maybe need update readme for the wheel part, I see it has been moved to pyproject.toxml last week

See #738

PR #740 (2024-09-24): manifest: Add submodule representation in as_dict
Submodules were missing when calling --freeze or --resolve

Fixes #739
closed 2024-09-24 by marc-hb