zephyrproject-rtos/west #742: Add progress info in west update command

yvesll, 2024-09-27:
HI, it would be helpful if we can know the progress when update lots of repos under a big project.
[tqdm](https://github.com/tqdm/tqdm) provides a beautiful bar but it seems not compatible with west's log module.
A simpler way like [1/99] is also good.

pdgendt, 2024-09-27:
This could go hand in hand with parallel updates, I want to tackle that in a future release.

I'll try and gather these issues in a `v2.0.0` milestone and look into those after the `v1.3.0` release.

marc-hb, 2024-09-27:
As discussed a bit in #301, git progress comes directly from git and `west` does not know anything about it (ignoring the fact the timing of git's progress bar can be jumping around as bad as the typical progress bar).

> This could go hand in hand with parallel updates, I want to tackle that in a future release.

BTW there's been a LOT of water under that bridge, here's one entry point:
- #713