zephyrproject-rtos/west #758: west build -p always -b nrf52840dk samples\basic\blinky fails

BettaSharma, 2024-10-21:
west build -p always -b nrf52840dk samples\basic\blinky is listed in 
https://docs.zephyrproject.org/latest/develop/getting_started/index.html

===============================================================================
Build the [Blinky](https://docs.zephyrproject.org/latest/samples/basic/blinky/README.html#blinky) with [west build](https://docs.zephyrproject.org/latest/develop/west/build-flash-debug.html#west-building), changing <your-board-name> appropriately for your board:

Ubuntu macOS Windows
cd %HOMEPATH%\zephyrproject\zephyr
west build -p always -b <your-board-name> samples\basic\blinky
===============================================================================

west boards lists

**nrf52840dk**

in the list of boards.

However, the command

**west build -p always -b nrf52840dk samples\basic\blinky**

results in

CMake Error at C:/Users/BSHARMA/Downloads/zephyrproject/zephyr/cmake/modules/boards.cmake:320 (message):
  Board qualifiers `` for board `nrf52840dk` not found.  Please specify a
  valid board target.

  **Valid board targets for nrf52840dk are:

  nrf52840dk/nrf52840

  nrf52840dk/nrf52811**

Call Stack (most recent call first):
  C:/Users/BSHARMA/Downloads/zephyrproject/zephyr/cmake/modules/zephyr_default.cmake:133 (include)
  C:/Users/BSHARMA/Downloads/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  C:/Users/BSHARMA/Downloads/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' -DWEST_PYTHON=C:/Users/BSHARMA/Downloads/zephyrproject/.venv/Scripts/python.exe '-BC:\Users\BSHARMA\Downloads\zephyrproject\zephyr\build' -GNinja -DBOARD=nrf52840dk '-SC:\Users\BSHARMA\Downloads\zephyrproject\zephyr\samples\basic\blinky'

However,

**west build -p always -b nrf52840dk/nrf52840 samples\basic\blinky**

succeeds!

That the boards name has to be **nrf52840dk/nrf52840** and NOT **nrf52840dk** has to be documented.

marc-hb, 2024-10-21:
Probably renamed in mega commit https://github.com/zephyrproject-rtos/zephyr/commit/8dc3f856229ce083c956aa301c31a23e65bd8cd8

marc-hb, 2024-10-21:
@BettaSharma can you please submit the rename?