zephyrproject-rtos/west #772: West doesn't find extension commands when run in a workspace without additional modules

kieranwlc, 2024-12-06:
**Describe the bug**

When running west within a workspace that only contains zephyr and no additional modules, west doesn't find the extension commands in /scripts/west_commands.yml. E.g. 'zephyr-export.'

**To Reproduce**

1. Initialise a workspace with a manifest that just contains zephyr and no additional modules:
E.g.
app/west.yml
```
manifest:
  remotes:
    - name: zephyrproject-rtos
      url-base: https://github.com/zephyrproject-rtos
  projects:
    - name: zephyr
      remote: zephyrproject-rtos
      revision: v4.0.0
```

```
west init -l app
west update
```
2. Run west or one of the west extension commands
`west help`
`west zephyr-export`

**Expected behavior**

West should be able to detect and run the extension commands in /zephyr

**Impact**

Annoyance

**Logs and console output**

```
west: unknown command "zephyr-export"; workspace {workspace_path} does not define this extension command -- try "west help" and "west -vv status"
```

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: just Zephyr
 - v4.0.0

pdgendt, 2024-12-06:
You didn't import anything from the project, but you could add the `west-commands` entry manually

```yaml
manifest:
  remotes:
    - name: zephyrproject-rtos
      url-base: https://github.com/zephyrproject-rtos
  projects:
    - name: zephyr
      remote: zephyrproject-rtos
      revision: v4.0.0
      west-commands: scripts/west-commands.yml
```

marc-hb, 2024-12-06:
> When running west within a workspace that only contains zephyr and no additional modules, west doesn't find the extension commands in /scripts/west_commands.yml.

The documentation could be clearer but I think extensions are not implicitly loaded:

https://docs.zephyrproject.org/4.0.0/develop/west/extensions.html

> The extension commands from Zephyr are added implicitly.

This looks more like a bug, @pdgendt maybe file it and close this one?

marc-hb, 2024-12-12:
Thanks @urob!

> but my interpretation is that using a [mapping keyword](https://docs.zephyrproject.org/latest/develop/west/manifest.html#option-3-mapping) like name-allowlist (or any of the other three import variants) merges that project's manifest with the parent manifest.

`name: zephyr; import: true` also imports `zephyr/scripts/west-commands.yml` through `zephyr/west.yml`. I just tested it. `import: false` (the default) does not.

> The mapping filter only restrict which projects are imported from zephyr/west.yml. Because the extensions are sourced in self, they are added when the manifest is imported regardless of any restrictions imposed by any of the mapping filters.

Yes, definitely looks like it!

> The documentation is a bit fuzzy 

I have not scanned every single line but it looks like there is indeed a documentation gap here.

It would also be good to check the source code to make sure this is all as intended but so far this all seems consistent.