zephyrproject-rtos/west #776: Multiple tests in `tests/test_config.py` are broken

carlescufi, 2025-01-01:
The following construct used pervasively in `tests/test_config.py`:

```
    with pytest.raises(subprocess.CalledProcessError) as e:
        cmd('...')
        assert 'error string' in str(e)
```

but this doesn't work for two reasons:

1. The `assert` statement must be outside the context manager (i.e. dedented one level) for it to execute at all if the exception is raised
2. The error string is very often not part of the exception object, so one needs to check the captured stderr instead

Something like this works instead:

```
    with pytest.raises(subprocess.CalledProcessError) as exc_info:
        cmd('...', stderr=subprocess.STDOUT)
    err_msg = exc_info.value.output.decode("utf-8")
    assert 'str' in err_msg
```

marc-hb, 2025-01-03:
Thanks @carlescufi ! Could you submit "quick" fixes for the "easy" ones? Leave the more complex ones for later / someone with more spare time. As far as test coverage is concerned, quantity matters!

marc-hb, 2025-01-03:
If you could add a `FIXME` next to the harder ones that would be the icing on the cake :-)

PR #777 (2025-01-10): tests: config: Fix assertions checks with pytest.raises
Dedent assertion checks as they were currently skipped, and make sure the correct error message is validated.

Fixes #776
closed 2025-01-10 by carlescufi