zephyrproject-rtos/west #782: west `--mr` does't work with commit hash

jmswu, 2025-02-09:
# The issue #
west option `--mr` doesn't work with commit hash

# How to reproduce #
* set up a new project by running `west init -m https://github.com/zephyrproject-rtos/zephyr --mr commit_hash`
* It will show this error
```sh
(.venv) ➜  zephyr-project-test west init -m https://github.com/zephyrproject-rtos/zephyr --mr 14fe06e4534c7f2c88e437a53bbb72998b516c4b
=== Initializing in /home/jms/projects/tmp/zephyr-project-test
--- Cloning manifest repository from https://github.com/zephyrproject-rtos/zephyr, rev. 14fe06e4534c7f2c88e437a53bbb72998b516c4b
Cloning into '/home/jms/projects/tmp/zephyr-project-test/.west/manifest-tmp'...
fatal: Remote branch 14fe06e4534c7f2c88e437a53bbb72998b516c4b not found in upstream origin
FATAL ERROR: command exited with status 128: git clone --branch 14fe06e4534c7f2c88e437a53bbb72998b516c4b https://github.com/zephyrproject-rtos/zephyr /home/jms/projects/tmp/zephyr-project-test/.west/manifest-tmp
```

# Environment #
* python version `Python 3.12.3`
* west version `West version: v1.3.0`
* OS:
```No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 24.04.1 LTS
Release:	24.04
Codename:	noble
```

pdgendt, 2025-02-09:
This is intentional, as the help description specifies for `--mr`: `manifest repository branch or tag name`.

You need to make a separate call in the `git checkout` directory for this.

jmswu, 2025-02-09:
Thanks for the reply, can I ask why? Or I should ask about this somewhere else?

pdgendt, 2025-02-10:
> Thanks for the reply, can I ask why? Or I should ask about this somewhere else?

I wasn't involved with the project when this was created, @mbolivar or @marc-hb can probably tell you more.

marc-hb, 2025-02-10:
It's quite complicated unfortunately. The explanation is scattered across various places; start from #522 and #531

closed 2025-02-10 by marc-hb

marc-hb, 2025-02-10:
Note: not every git server allows fetching SHA1s directly. 
See `uploadpack.allowTipSHA1InWant`, `uploadpack.allowReachableSHA1InWant`,  `uploadpack.allowAnySHA1InWant` (https://git-scm.com/docs/gitprotocol-capabilities.html#_allow_tip_sha1_in_want)

Garbage collecting something from git is very hard because of the `reflog` and  other things. So, when a server allows direct SHA1 fetching, it is much harder to make "accidents" unreachable.

This is slightly off-topic but worth knowing.