zephyrproject-rtos/west #785: Support `uv` as an alternative manager to `pip` for package management

cdwilson, 2025-03-17:
`uv` (https://github.com/astral-sh/uv) is becoming a popular tool for creating and managing Python virtual environments (mainly due to the fact that it's really fast).

It's currently possible to use `uv pip` to install Python dependencies for Zephyr, e.g.:

```sh
west packages pip | xargs uv pip install
```

It would be nice if `uv pip` could be added as a supported "manager" for the `west packages` command so that `uv` would be used under the hood to install packages.

marc-hb, 2025-03-17:
Thanks @cdwilson! `west packages` was designed to be open to other managers so this sounds like a good idea but... considering that:
- `west packages` has only two features now: list and install
- listing seems already compatible with `uv`
- The simple one-liner you shared provides installation compatibility with `uv` for a very "low-cost".
- `west packages` is not a frequently used command.

... this all seems to concur to make this good idea... with very low priority right now? Unless I'm missing something? `uv` is brand new to me (thanks for sharing!) so I might missing something.

pdgendt, 2025-03-18:
In any case, `west packages` is a Zephyr extension, so this issue should be moved to the Zephyr project.