zephyrproject-rtos/west #795: please get rid of west

paigeadelethompson, 2025-03-29:
I've never actually been able to get it to work or do anything meaningful anywhere that I'm supposed to use it.

paigeadelethompson, 2025-03-29:
Like seriously I'm about throw this nrf 7002 and 52840 in the trash because they both fill me with disgust every time I look at them, because of this stupid program

pdgendt, 2025-03-29:
Similar to https://github.com/zephyrproject-rtos/zephyr/issues/87844 this isn't constructive feedback in any way or form. There are many community members that would gladly help, but not like this.

closed 2025-03-29 by pdgendt

marc-hb, 2025-03-29:
Last time I checked, `west` was actually optional. Of course it made things harder because you have to do things "manually" instead but it was possible.

Considering not just the tone, but also the complete lack of content in the description, I suspect the usability issues experienced were not actually related to `west`. Because it's the front-end hence "messenger", neither the first time  nor the last time that issues unrelated to west are filed in the west tracker.

paigeadelethompson, 2025-03-31:
Dude it could be anything if it just worked I DON'T CARE MAN. Do you have any idea how much of a pain in the a** it was to make this stupid script work: https://gist.github.com/paigeadelethompson/f83661c77bd1d56b1c388fff5821b426

and then it wouldn't even work right I had to dump it into some SDK directory to use it. Look dude all's I'm saying is, I don't even want to mess with zephyr if I can avoid it but only because it's damn near impossible to just set it up and build I have to have 235789283904098 copies of the SDK everywhere, and probably a bunch of stuff in my env to go with all of the other crap that I have in my env and then it just doesn't even work. I really wouldn't care what it was if it worked, it could be gentoo's emerge command that'd be fine if it did what it was supposed to do.

paigeadelethompson, 2025-03-31:
sorry if that hurts your feelings (oh my god) but its the truth dude, if I were like 16 or 17 still I'd have smashed 10 or so keyboards over it

pdgendt, 2025-03-31:
This issue was created after things already went south, there was no information posted in the initial description for people to able to help.

Please [read the documentation](https://docs.zephyrproject.org/latest/develop/getting_started/index.html#asking-for-help) on how to get help, if you'd like to continue.

paigeadelethompson, 2025-04-02:
```
➜  ~ cd /home/netcraveos/.oh-my-sdk/dist/nrf/zephyr/samples/bluetooth/hci_usb   
ℹ  OH_MY_SDK_DIST value: /home/netcraveos/.oh-my-sdk/dist
ℹ  Looking for NRF directory at: /home/netcraveos/.oh-my-sdk/dist/nrf
ℹ  Looking for Command Line Tools at: /home/netcraveos/.oh-my-sdk/dist/nrf-command-line-tools
ℹ  Activating Python virtual environment...
ℹ  Looking for virtual environment at: /home/netcraveos/.oh-my-sdk/pyenv/nrf
ℹ  Exporting Zephyr CMake package...
Zephyr (/home/netcraveos/.oh-my-sdk/dist/nrf/zephyr/share/zephyr-package/cmake)
has been added to the user package registry in:
~/.cmake/packages/Zephyr

ZephyrUnittest (/home/netcraveos/.oh-my-sdk/dist/nrf/zephyr/share/zephyrunittest-package/cmake)
has been added to the user package registry in:
~/.cmake/packages/ZephyrUnittest

✅ NRF Connect environment activated!

ℹ  Available commands:
  • west help                  Show west command help
  • nrfutil help              Show nrfutil command help
  • omsdk deactivate          Deactivate environment
(nrf) ➜  hci_usb git:(9ec6585407f) ✗ west build --board=nrf7002dk/nrf5340/cpuapp -c                          
-- west build: generating a build system
Loading Zephyr module(s) (Zephyr base (cached)): sysbuild_default
-- Cache files will be written to: /home/netcraveos/.cache/zephyr
-- Found west (found suitable version "1.3.0", minimum required is "0.14.0")
-- Board: nrf7002dk, qualifiers: nrf5340/cpuapp
Parsing /home/netcraveos/.oh-my-sdk/dist/nrf/zephyr/share/sysbuild/Kconfig
Loaded configuration '/home/netcraveos/.oh-my-sdk/dist/nrf/zephyr/samples/bluetooth/hci_usb/build/zephyr/.config'
No change to configuration in '/home/netcraveos/.oh-my-sdk/dist/nrf/zephyr/samples/bluetooth/hci_usb/build/zephyr/.config'
No change to Kconfig header in '/home/netcraveos/.oh-my-sdk/dist/nrf/zephyr/samples/bluetooth/hci_usb/build/_sysbuild/autoconf.h'
-- 
   *****************************
   * Running CMake for hci_usb *
   *****************************

Loading Zephyr default modules (Zephyr base (cached)).
-- Application: /home/netcraveos/.oh-my-sdk/dist/nrf/zephyr/samples/bluetooth/hci_usb
-- CMake version: 3.31.6
-- Cache files will be written to: /home/netcraveos/.cache/zephyr
-- Zephyr version: 4.0.99 (/home/netcraveos/.oh-my-sdk/dist/nrf/zephyr)
-- Board: nrf7002dk, qualifiers: nrf5340/cpuapp
-- Found host-tools: zephyr 0.17.0 (/home/netcraveos/zephyr-sdk-0.17.0)
-- Found toolchain: zephyr 0.17.0 (/home/netcraveos/zephyr-sdk-0.17.0)
-- Found BOARD.dts: /home/netcraveos/.oh-my-sdk/dist/nrf/zephyr/boards/nordic/nrf7002dk/nrf7002dk_nrf5340_cpuapp.dts
-- Generated zephyr.dts: /home/netcraveos/.oh-my-sdk/dist/nrf/zephyr/samples/bluetooth/hci_usb/build/hci_usb/zephyr/zephyr.dts
-- Generated pickled edt: /home/netcraveos/.oh-my-sdk/dist/nrf/zephyr/samples/bluetooth/hci_usb/build/hci_usb/zephyr/edt.pickle
-- Generated zephyr.dts: /home/netcraveos/.oh-my-sdk/dist/nrf/zephyr/samples/bluetooth/hci_usb/build/hci_usb/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/netcraveos/.oh-my-sdk/dist/nrf/zephyr/samples/bluetooth/hci_usb/build/hci_usb/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: /home/netcraveos/.oh-my-sdk/dist/nrf/zephyr/samples/bluetooth/hci_usb/build/hci_usb/zephyr/dts.cmake

warning: Deprecated symbol DEPRECATED_UART_NRFX_UARTE_LEGACY_SHIM is enabled.

/home/netcraveos/.oh-my-sdk/dist/nrf/zephyr/samples/bluetooth/hci_usb/build/hci_usb/zephyr/.config.sysbuild:2: warning: attempt to assign the value 'y' to the undefined symbol PARTITION_MANAGER_ENABLED

/home/netcraveos/.oh-my-sdk/dist/nrf/zephyr/samples/bluetooth/hci_usb/build/hci_usb/zephyr/.config.sysbuild:6: warning: attempt to assign the value 'n' to the undefined symbol CHIP

warning: HAS_NORDIC_DRIVERS (defined at modules/hal_nordic/Kconfig:7) has direct dependencies 0 with value n, but is currently being y-selected by the following symbols:
 - SOC_SERIES_NRF53X (defined at soc/nordic/Kconfig.soc:36, soc/nordic/nrf53/Kconfig.soc:6, soc/nordic/nrf53/Kconfig:6), with value y, direct dependencies y (value: y), and select condition SOC_FAMILY_NORDIC_NRF (value: y)

warning: HAS_CMSIS_CORE (defined at modules/cmsis/Kconfig:7) has direct dependencies 0 with value n, but is currently being y-selected by the following symbols:
 - CPU_CORTEX_M (defined at arch/arm/core/Kconfig:6), with value y, direct dependencies ARM (value: y), and select condition ARM (value: y)

warning: CMSIS_CORE_HAS_SYSTEM_CORE_CLOCK (defined at soc/nxp/imxrt/imxrt6xx/Kconfig.defconfig:8, modules/cmsis/Kconfig:31) has direct dependencies (SOC_SERIES_IMXRT6XX && SOC_FAMILY_NXP_IMXRT) || (HAS_CMSIS_CORE && 0) with value n, but is currently being y-selected by the following symbols:
 - SOC_FAMILY_NORDIC_NRF (defined at soc/nordic/Kconfig.soc:19, soc/nordic/Kconfig:9), with value y, direct dependencies y (value: y), and select condition ARM (value: y)

warning: HAS_NRFX (defined at modules/hal_nordic/nrfx/Kconfig:4) has direct dependencies 0 with value n, but is currently being y-selected by the following symbols:
 - SOC_SERIES_NRF53X (defined at soc/nordic/Kconfig.soc:36, soc/nordic/nrf53/Kconfig.soc:6, soc/nordic/nrf53/Kconfig:6), with value y, direct dependencies y (value: y), and select condition SOC_FAMILY_NORDIC_NRF (value: y)

[... 31 lines cut ...]
  /home/netcraveos/.oh-my-sdk/dist/nrf/zephyr/cmake/modules/zephyr_default.cmake:133 (include)
[... cut ...]
  /home/netcraveos/.oh-my-sdk/dist/nrf/zephyr/share/sysbuild-package/cmake/SysbuildConfig.cmake:8 (include)
  template/CMakeLists.txt:10 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/home/netcraveos/.oh-my-sdk/dist/nrf/zephyr/samples/bluetooth/hci_usb/build -GNinja -S/home/netcraveos/.oh-my-sdk/dist/nrf/zephyr/share/sysbuild -DAPP_DIR:PATH=/home/netcraveos/.oh-my-sdk/dist/nrf/zephyr/samples/bluetooth/hci_usb
(nrf) ➜  hci_usb git:(9ec6585407f) ✗ 

```
😂

kartben, 2025-04-03:
Hi @paigeadelethompson  -- I would like to remind you the Zephyr Project does have a [Code of Conduct](https://github.com/zephyrproject-rtos/zephyr/blob/main/CODE_OF_CONDUCT.md), and your behaviour here (as well as https://github.com/zephyrproject-rtos/zephyr/issues/87844) is completely inappropriate and certainly not the best way for you to get actual help from other collaborators on the project. 
If you would like to take a moment to provide feedback that goes beyond "this thing doesn't work" or "get rid of that", I am sure you will find many people willing to help and have a discussion :) And it looks like your oh-my-sdk tool might be a step in the right direction.

Thank you.

paigeadelethompson, 2025-04-07:
yeah you're right I'm sorry, just seems really passive aggressive and it's frustrating. I mean sure, setting up the SDK is easy: 

- https://github.com/Ravenhammer-Research/oh-my-sdk/blob/master/nrf/_oh_my_sdk_install_nrf.zsh
- https://github.com/Ravenhammer-Research/oh-my-sdk/blob/master/nrf/_oh_my_sdk_install_jlink.zsh
- https://github.com/Ravenhammer-Research/oh-my-sdk/blob/master/nrf/_oh_my_sdk_create_venv.zsh
- https://github.com/Ravenhammer-Research/oh-my-sdk/blob/master/nrf/_oh_my_sdk_install_zephyr_sdk.zsh

Except it's a lot and it's spread out all over the place and so it's not. I mean it has the hallmarks of being something that somebody actually cares about.. but then the source of truth leads you to more disappointment and nothing gets done and another hour is gone. People would have me believe Zephyr is awesome (it actually is I think it's great) but that is perhaps assuming you're working exclusively with vscode extensions.

Here's the other thing, and this is not really a Zephyr problem but I didn't even know that they had another nrfutil (besides the PyPI): 

https://github.com/Ravenhammer-Research/oh-my-sdk/blob/master/nrf/_oh_my_sdk_install_nrf.zsh#L73

Like this tool probably does everything I need it to do but I didn't even find it until I was finishing up this zsh plugin and trying to get the pip dependencies sorted out for the PyPI module. I don't think that PyPI module is maintained by anybody to be honest, and the tool that I discovered by accident is definitely the right one.  

No that's not a Zephyr problem but at the same time how did their ecosystem end up the way that is? It seems they started with Zephyr as near as I can so it ends up being something like my first programming gig working with Codewarrior for PalmOS and only being able to use codewarrior because of the Symbol library that i had work with; gcc was a much better option than a pirated copy of Windows XP / pirated copy of Codewarrior but that's the way things were.. like 20 years ago. 

For me it's kinda like when you work for a poor company, imagine it, and you find a local meetup for some automation software that is "opensource" and everyone says its great and maybe it is...except everybody at the meetup works for a company that is actually paying for a support contract and that's the part they don't tell you. That's kinda how this makes me feel, dude I can't help it-- people do this and the will even go as far as to try to gaslight you if you call them out for it. I'm not saying that's true here, I just feel like for as little effort as I put into to making something that reproduces the tools that are required to use this somebody could have done it better. Like I don't know, somebody who really cares enough about the project to be associated with it officially? It just seems like if that were really the case, that somebody would want people who just wanna use actually be able to.