zephyrproject-rtos/west #796: Release 1.4.0

pdgendt, 2025-03-31:
An issue to track progress on the next (v1.4.0) release. (See [MAINTAINERS.rst](https://github.com/zephyrproject-rtos/west/blob/c399c012570c51ae806fdb5bb940d7ab1bd1c3d2/MAINTAINERS.rst) for details)

# Pre-release test plan

- [ ] Fork `v1.4-branch` release branch
- [ ] Update version for `main` to `v1.4.99`
- [ ] Update version in `pyproject.toml`
  - [ ] `v1.4.0a1`
- [ ] Run `tox` using docker
- [ ] Build `v1.4.0a1`
- [ ] Upload `v1.4.0a1` to `pypi`
- [ ] Validate `v1.4.0a1` for Zephyr workspace on
  - [ ] Windows
  - [ ] macOS
  - [ ] Linux

# Release plan

- [ ] Update version in `pyproject.toml` to `v1.4.0`
- [ ] Tag release
- [ ] Create GH release
- [ ] Upload `v1.4.0` to `pypi`
- [ ] Send email to Zephyr lists

kartben, 2025-03-31:
+ Publish release notes on Zephyr docs 
https://docs.zephyrproject.org/latest/develop/west/release-notes.html
+ Update API docs https://docs.zephyrproject.org/latest/develop/west/west-apis.html

mbolivar, 2025-03-31:
Probably a good idea to make sure that the API docs build cleanly **before** tagging and uploading the release.

Sometimes, syntax errors in the pydoc require code changes before we can update the docs build due to the use of sphinx autodoc features.

mbolivar, 2025-03-31:
@pdgendt can you clarify the purpose of this issue compared to the process in the MAINTAINERS file in the repository? I think it'd be best to keep any release plans in sync with the MAINTAINERS doc, even if that requires some updates to that doc

pdgendt, 2025-04-01:
> Probably a good idea to make sure that the API docs build cleanly **before** tagging and uploading the release.

Right, changed the order.

> [@pdgendt](https://github.com/pdgendt?rgh-link-date=2025-03-31T22%3A50%3A43.000Z) can you clarify the purpose of this issue compared to the process in the MAINTAINERS file in the repository? 

I took the inspiration from the [Zephyr release checklist](https://github.com/zephyrproject-rtos/zephyr/issues/86867) issues, creating awareness of the upcoming release, and allow for people to comment on the process/progress.

> I think it'd be best to keep any release plans in sync with the MAINTAINERS doc, even if that requires some updates to that doc

I'll make a PR to update the docs.

pdgendt, 2025-04-01:
@mbolivar I want to document the process of updating the docs, but I can't find the instructions. Sorry for my ignorant question, this was clearly missed in https://github.com/zephyrproject-rtos/west/issues/735 as well.

mbolivar, 2025-04-02:
Thank you so much for taking this on. I really appreciate you stepping up.

I'm not sure I ever documented the documentation update process, but basically it goes like this:

1. Scan the git log for major changes
2. update https://docs.zephyrproject.org/latest/develop/west/release-notes.html following the pattern in the document for previous releases based on step 1
3. update https://docs.zephyrproject.org/latest/develop/west/west-apis.html#west-apis based on step 1; this can require some manual testing and west code changes, based on step 1
4. update any other major pages based on the changes in the release. For example, if new manifest syntax changes were merged, update the schema information in https://docs.zephyrproject.org/latest/develop/west/manifest.html#version

For step 3, it basically works like this:

3.1. Check out and install your "alpha" west release locally (where by alpha I mean it in the sense of https://peps.python.org/pep-0440/#pre-releases)
3.2. Update west-apis.rst in the zephyr repository accordingly based on what changed since last time and build the docs
3.3. if it builds, read the results and make sure it's ok. If so, you are done!
3.4. if it doesn't build or some pydoc content in the west repo has problems based on review, fix as needed in the west repository, try again locally until you're satisfied, and get the changes merged into mainline west

I would go through the trouble of actually cutting alpha releases on pypi for the record and reproducibility. You can see that in this example PR:

https://github.com/zephyrproject-rtos/zephyr/pull/55009

But feel free to remix this process as you see fit as the maintainer, of course. The main thing I want to stress is that you should make sure the docs are building OK before you cut any non-alpha release tags, or otherwise you won't be able to document that release!

mbolivar, 2025-04-02:
@pdgendt when announcing the release, I would send email to at least the announce list following the pattern in https://lists.zephyrproject.org/g/announce/topic/west_v1_0_0/97096234

closed 2025-05-23 by pdgendt