zephyrproject-rtos/west #799: Make a proper dedicated top-level documentation site

yashi, 2025-04-02:
https://discord.com/channels/720317445772017664/733037524498382881/1356918374231445524

Make a proper West document and host it by itself. We can based on the current West sections in the Zephyr Doc.

Once we get it up running, we can start trimming the West section in the Zephyr doc or replace it with links to West doc.

marc-hb, 2025-04-02:
That would be great but please edit the description and add a summary of the Discord discussion which is NOT available unless you have a Discord account (reproduce with incognito mode). Being a project member may be required too.

marc-hb, 2026-11-12:
Redirecting the conversation from https://github.com/zephyrproject-rtos/west/pull/860#discussion_r4551011332

> `west completion` is for both West built-ins, and Zephyr-specific west extensions. 
>  The problem is that the `west completion` extension command does live only in Zephyr.

I forgot about that extension... It's a bit pointless TBH: the only thing it does is hiding the location of the actual file. So the doc tells the user to redirect its output to... a file, which is just a pretty convoluted way to make a file copy.

>  The problem is that the west completion extension command does live only in Zephyr.

I think the easiest in the longer term is to split the actual completion files in two and "de-obfuscate" the `west extension` command in Zephyr by having it print the sample copy command to run - with the location of the file (half) to copy. The bare `west` does not need the same command.