zephyrproject-rtos/west #807: Deprecate `pyKwalify` for YAML validation

pdgendt, 2025-04-15:
The [pyKwalify](https://pypi.org/project/pykwalify/) project hasn't received updates since 2020 and the highest supported Python version is 3.9.

As an alternative, I propose moving to [JSON schema](https://json-schema.org/)

- An active project
- [YAML as a superset of JSON](https://json-schema-everywhere.github.io/yaml)
- Clear [specifications](https://json-schema.org/specification)
- Backed by large companies

marc-hb, 2025-04-16:
Thanks for spotting and filing this! Do you have a rough idea of the extend of pyKwalify usage currently? I mean everywhere it's used for in `west`.

How different are the APIs?

pdgendt, 2025-04-17:
> Thanks for spotting and filing this! Do you have a rough idea of the extend of pyKwalify usage currently? I mean everywhere it's used for in `west`.

It's used to validate manifest and command extension files. The changes required are limited, but as the schema is versioned in West, it might make sense to bump from currently 1.2 to 2.0? Even though it should be backwards compatible.

> How different are the APIs?

The API is almost identical, there's a single `validate` function that throws an error if it's malformed. However the schema syntax is different.

Hoohaha, 2026-01-07:
JSON Schema is lighter, more flexible than pyKwalify, and widely adopted across the software industry. In my opinion, we should switch to JSON Schema.

pdgendt, 2026-01-07:
Yes, I need to get back to this. I created a WIP branch https://github.com/zephyrproject-rtos/west/compare/main...pdgendt:west:json-schema a while ago, will try to update it and open a PR soon.