zephyrproject-rtos/west #823: schema: Document

yashi, 2025-07-02:
West v0.10 says that it allowed a manifest without any project as you can see the commit 984aec8ecdc9b19fbace7707f9fd5ddacc99d309, the current version errors out because of:

```python
    data = as_dict['manifest']

    # Make sure this version of west can load this manifest data.
    # This has to happen before the schema check -- later schemas
    # may incompatibly extend this one.
    if 'version' in data:
```

When the manifest file doens't have anything except `manifest:`, the `data` is None.  testing None with `in` raises an exception

```
❯ python3
Python 3.13.3 (main, Apr 10 2025, 21:38:51) [GCC 14.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> if 'hoge' in None:
...     pass
... 
Traceback (most recent call last):
  File "<python-input-0>", line 1, in <module>
    if 'hoge' in None:
       ^^^^^^^^^^^^^^
TypeError: argument of type 'NoneType' is not iterable
```

marc-hb, 2025-07-02:
I took a closer look. The test passes because it creates an empty manifest but it does not go as as far as running `west update`.

@yashi can you explain what is your use case for downloading nothing?

If there is a simple fix then please submit, otherwise I'm tempted to close this as "wontfix".

yashi, 2025-07-02:
Actually, I don’t have a use case yet. My intention is to work on issue #246.  I was just reviewing the schema and I noticed a few inconsistencies.

A simple document fix is OK for me for now. But if maintainers want to do in any otherway, I can work on it.

yashi, 2025-07-03:
Just to be clear.  These error occurs when I run `west manifest --validate` right after `west init`.

**manifest: only**
```
❯ rm -rf app .west
❯ mkdir app
❯ echo manifest: > app/west.yml
❯ west init -l app             
=== Initializing from existing manifest repository app
--- Creating /tmp/asdf/.west and local configuration file
=== Initialized. Now run "west update" inside /tmp/asdf.
❯ west manifest --validate
FATAL ERROR: can't run west manifest; it requires the manifest, which was not available. Try "west -vv manifest --validate" to debug.
```

**manifest: with empty projects:**
```
❯ rm -rf app .west
❯ mkdir app
❯ printf 'manifest:\n  projects:' > app/west.yml
❯ west init -l app
=== Initializing from existing manifest repository app
--- Creating /tmp/asdf/.west and local configuration file
=== Initialized. Now run "west update" inside /tmp/asdf.
❯ west manifest --validate 
Traceback (most recent call last):
  File "/tmp/asdf/.venv/bin/west", line 8, in <module>
    sys.exit(main())
             ~~~~^^
  File "/home/yashi/src/west/src/west/app/main.py", line 1199, in main
    app.run(argv or sys.argv[1:])
    ~~~~~~~^^^^^^^^^^^^^^^^^^^^^^
  File "/home/yashi/src/west/src/west/app/main.py", line 269, in run
    self.load_manifest()
    ~~~~~~~~~~~~~~~~~~^^
  File "/home/yashi/src/west/src/west/app/main.py", line 288, in load_manifest
    self.manifest = Manifest.from_topdir(topdir=self.topdir,
                    ~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^
                                         config=self.config)
                                         ^^^^^^^^^^^^^^^^^^^
  File "/home/yashi/src/west/src/west/manifest.py", line 1231, in from_topdir
    return Manifest(topdir=topdir, config=config,
                    importer=importer, import_flags=import_flags)
  File "/home/yashi/src/west/src/west/manifest.py", line 1534, in __init__
    self._load_validated()
    ~~~~~~~~~~~~~~~~~~~~^^
  File "/home/yashi/src/west/src/west/manifest.py", line 2014, in _load_validated
    self._load_projects(manifest_data, url_bases, defaults)
    ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/yashi/src/west/src/west/manifest.py", line 2305, in _load_projects
    for pd in manifest['projects']:
              ~~~~~~~~^^^^^^^^^^^^
TypeError: 'NoneType' object is not iterable
```

marc-hb, 2025-07-11:
Fix submitted in #825