zephyrproject-rtos/west #828: Cache for initial "west update" in CI

thorsten-klein, 2025-07-14:
### Summary

Hello,

we are currently setting up our first cloud-based CI build for our zephyr project.

In our initial runs, we’ve observed the following timings:

Note: It is a freshly started VM in CI.

- Cloning our repository: **1 second**
- Setting up the initial pip3 environment (`pip3 install -r requirements.txt`): **15 seconds**
- `west update`: **90 seconds**
- `west packages pip --install`: **15 seconds**
- `west build`: **30 seconds**

As you can see, the `west update` step is by far the most time-consuming and generates significant network traffic (because of the `git clone` commands which are run). We would like to optimize this by introducing caching.

Do you already have a recommended solution for caching `west update`?

### Describe the solution you'd like

One idea we’re considering is to cache the Git repositories that west clones. For example:

- Keep a centralized mirror of the modules in a local directory (e.g., `~/.west/git`)
- Configure west to clone from these local mirrors instead of remote URLs
  - This could be done using `git worktree`, or
  - By cloning from a local `--mirror` repository using file paths

This would allow the `west update` step to pull from the local system, significantly reducing clone time and external traffic.

### Alternatives

Are there already alternative solutions existing?

thorsten-klein, 2025-07-14:
Thank you for your prompt response.

I have tested the process, and I expected that setting `--name-cache NAME_CACHE` would automatically clone the repositories to the specified location on the first run. Unfortunately, this does not happen.

It appears that I need to manually clone the repositories to that location beforehand. This is quite challenging for me, especially since most of the repositories are transitive dependencies.

marc-hb, 2025-07-15:
> If I find the time, I will explore the possibility of adapting west to support a new flag --auto-cache. This flag could be used in conjunction with --name-cache or --path-cache to automatically clone repositories to the specified path (where they were searched but not found yet). 

That looks like a simple and great user interface and I'm sure a lot of people would love this feature but I'm worried about (at least) two things:
- Corner cases and test coverage. Notably: how would this interact with manifest changes and especially imports which tend to make things very "dynamic". 
- Complexity and long-term maintenance. In other words, what happens after you walk away? :-) Typical issue with open source.

BTW I would initialize these as bare repos if they don't exist yet to avoid accidental usage.

> The repository would then be fetched and copied to its final location instead of being fully cloned.

I'm a bit confused by this, maybe because I'm not familiar with the existing cache feature. Are you suggesting some user-visible, behavior change here compared to what is already there? I mean besides refreshing the cache (which should be barely visible).

> > It appears that I need to manually clone the repositories to that location beforehand.

> Using west for that shouldn't be too hard, is it?

I'm afraid you missed this bit. I see this sort of "manual" cache initialization as a potentially useful stepping stone / prototyping step. I mean: if `west` can't even perform this with two commands yet, then it's pretty far from performing it in a single, future, `--auto-cache` command! Using these two steps "in anger" in CI should provide useful experience. This is also be a good excuse to get something like `--fetch-opt=--bare` to work; you get the idea.

thorsten-klein, 2025-07-16:
Hello @marc-hb,

I've put together an initial proposal for the `--auto-cache` flag. I think this helps to avoid misunderstanding.
I'd love to hear your thoughts on it:
https://github.com/zephyrproject-rtos/west/compare/main...thorsten-klein:west:issue-828-auto-cache?expand=1

Once I receive your feedback and if you like it, I'll proceed with creating the official Pull Request which can be officially reviewed.

Thank you!

marc-hb, 2025-07-25:
> but I'm worried about (at least) two things:
>
> Corner cases and test coverage. Notably: how would this interact with ...

Ta-da: git submodules have just entered the room:
- https://github.com/zephyrproject-rtos/west/pull/829#issuecomment-3116875431